zerog2k/stc_diyclock #3: Other kit models

wSmithyeah, 2016-05-25:
Hi

Firstly, this repository is really cool. Fantastic work.

I have a [similar ebay clock](http://www.ebay.co.uk/itm/DIY-kit-LED-electronic-clock-microcontroller-LED-digital-watch-time-thermometer-/181846168677?var=&hash=item2a56e05465:m:m63ylfNLE0G8Dgu0PJdruZw)

They have the same micro controller, thermistor and photo resistor.
Would this firmware also work with it?
Can I use this [stc programmer](http://www.ebay.co.uk/itm/231897030676) to write the firmware, or is it not required?
Sorry for all the questions - I'm new to STC

Best regards, Warren

zerog2k, 2016-05-26:
Hi Warren.
If you are using STC15Fxxxx series, you dont need a separate "programmer" as this already has a built-in serial boot loader firmware - you program through the serial port of the chip (as documented towards the end of the data sheet). The idea is similar to arduino if you are familiar with it. You only need a cheap usb to uart adapter for this and either use the open-source stcgal or oem provided windows app, STC-ISP for programming over serial port.

As to programming it, I can only suggest reading the documentation links, especially the stc datasheet, and maybe sdcc user guide (if you want to dive deeper into the compiler/assembler/linker details).

It's actually pretty simple - the biggest challenge is (re)writing a "complete" functionally-equivalent firmware in C which fits in 4k of code-space, as opposed to in assembly (which would probably be easier to fit, but harder to code, follow, and maintain.)

Regarding the compatibility with your kit, I took a look at the ebay listing but it did not have a close-up of the board nor a schematic to compare to. It does look similar but from the photo, assuming thats the same as the one you have, there might be slight differences.
If they provided a documentation page, hopefully it had the schematics and board layout. Compare to the schematics of the board I developed for:
https://github.com/zerog2k/stc_diyclock/blob/master/docs/DIY_LED_Clock.png

If they are not the same, it's probably trivial to make some minor changes to this to get it to run on your kit.

Good luck.

wSmithyeah, 2016-06-01:
Hi Zerog2k

That was a great answer. Many thanks. I'll give the reprogramming it a try. If I succeed then I'll take pictures and maybe you'll want to add this clock to your project too.

I'm not on Windows, just Linux, but it looks like stcgal is Python based, so will work on Linux.

The only part of the clock I wanted to change is the brightness of the display - so that it dims when there is less light and visa versa. 

I'm pretty good with code, I used to lecture in c and cpp, but rubbish with electronics. If it doesn't work I'll do some digging and maybe the changes can be adapted at the pre compiler stages for the differences between the clocks.

I'll let you know.
Best regards
  Warren

closed 2016-09-30 by zerog2k