zerog2k/stc_diyclock #5: Windows Software

Primus007, 2016-06-22:
please can you or any other people windows software to compile and flash ??
a little how to for windows users would be nice too :+1: 

many thanks for your work!

zerog2k, 2016-06-22:
I only have mac and linux at the moment to develop on.
But in theory for compilation sdcc is available for windows (might need a few makefile tweaks for windows paths, etc), and you can use the oem software (stc-isp) for flashing.

Primus007, 2016-06-22:
thank you.
sdcc i a command line tool. anyone knows a gui for this?

currently i try Keil c51 to code/compile and make a hex file.

stc-isp i have downloaded. i have selected the STC15F204EA mcu. **must i do some settings to work correctly?**
keil is missing some includes like stc12,stdint,stdio, **where can i get it?**

do you think i can use this module to connect to the clock P3.0 TX and P2.0 RX pin? 
http://www.aliexpress.com/item/FT232RL-FTDI-USB-to-TTL-Serial-Adapter-Module-for-Arduino-Mini-Port-3-3V-5V/2043815349.html

zerog2k, 2016-06-22:
You might be able to use keil, but several things would have to be changed slightly. Most of the includes, e.g. stc12.h, etc. are written for sdcc. I think you also have to import the CDB for STC chips into keil (I think the stc-isp program provides this). There are a few differences here and there. Yes, sdcc is cli, but I guess you can incorporate it into an IDE of your choosing, e.g. netbeans, eclipse, etc - but that's a different topic altogether. 

Regarding flash programming, it's always done over serial uart using the ROM bootloader onboard STC chips. On the clock kit, I soldered a header onto J1 (TXD/RXD/GND/VCC), and also pulled the leg of RST (pin 18?) up from socket and have that tied to a jumper which is hooked up to DTR pin (for managing reset/boot sequence in stcgal for programming convenience). I'm not sure if stc-isp uses dtr for reset into bootloader. May need to manually powercycle (vcc or gnd) when attempting download.

Primus007, 2016-06-23:
thanks for your answer!
> I think you also have to import the CDB for STC chips into keil
yes that is right. but i cant find a CDB with the STC15F204EA. But i find a STC15F204EA.H file. i thinl this is ok also.

thanks for your help. i will try it in the next days and hope all gets well. i want to make me a special version without the cycling.

**is it possible to get the original code from the clock?** anywhere in the www to download?

Primus007, 2016-06-28:
ok i have now the software for windows "micro-ide" with sdd integrated. All works but there is a Problem by compiling ds1302.c in three lines.
see the attachment.
please can you help me? thanks!
[image]

zerog2k, 2016-06-28:
are you compiling w/ sdcc or keil?

(@jjmz did some code size optimization - which is highly "sdcc", e.g. calling pragmas, etc)

The line seems like valid C for right shifting... It's not immediately obvious what might be going on.

jjmz, 2016-06-28:
Sorry if I broke something. That's true I am only using sdcc.
Thus I hesitate issuing a PR, since my latest code is really "sdcc" & optimized... (2278 bytes, with seconds display & zeroing, MMDD/DDMM option for date display) but the code now greatly diverges from yours.

zerog2k, 2016-06-28:
@jjmz no worries - this is a good thing... 
@Primus007 we can make sdcc vs keil preprocessor macro conditionals - this is how some SiLabs IDE MCU code (C8051F320 for example) does to manage multiple compilers. They use a compiler_defs.h to abstract and normalize compiler differences...

I see stuff like:
```
#if defined SDCC

#if (SDCC >= 300)
...

// Keil C51
// http://www.keil.com
#elif defined __C51__
...

```

You can see the full example of what they are doing here:
https://gist.github.com/zerog2k/778af78a77c956eaafef94948853ef7e

Primus007, 2016-06-28:
currently i use this: Micro-IDE SDCC 8051 Development System http://www.bipom.com/products/us/319513.html
there is SDCC 2.9.0 included.
But i have the header files from the newest sdcc version for windows x64 v3.6.0

With Keil are more errors. with sdcc only the one seen in the screenshot.

zerog2k, 2016-06-28:
yeah, sdcc < 3.0 is old. This was developed on 3.4 and 3.5. Latest is 3.6.
I dont know anything about that IDE, but maybe you can re-point it to the newer sdcc toolchain. I'm guessing that will fix this particular issue.

Primus007, 2016-06-28:
**thank you very much.** it seems to work! I have now the .HEX file.

Can i make a Backup from the current Clock before i upload the new .HEX or is the original .HEX anywhere to download?

zerog2k, 2016-06-28:
regarding that error and your screenshot above, maybe you need to reset and resync to git... Looks like you are missing a line from the code base:
https://github.com/zerog2k/stc_diyclock/blob/master/src/ds1302.c#L123

regarding backup - short answer - there is none. Long answer:
1. this is a security "feature" of stc bootloader that you cannot download code from flash of a programmed mcu, and
2. I have not seen anyone release the firmware (even in bin/hex) format of the original, although I suspect there might be some minor variations between kits in the wild.

closed 2016-09-30 by zerog2k