zerog2k/stc_diyclock #8: STC15W408AS based clock

hachi, 2016-08-11:
I purchased http://www.ebay.com/itm/152077409965 and on a whim decided to load this firmware onto it, which has turned out badly for the device.

I can't find a datasheet for this processor, so this might be a lost cause, would you mind helping me decide?

The clock starts up and for about 100ms shows all digits fully lit, then the display shifts to "1F:7F." with the 'colon' blinking at a rate about 2x faster than it should be. No button presses or combinations can trigger a result as far as I can tell. The light sensor works and I can see it visibly dim the display in darkness.

I have injected different display tests using led.h bit masks to test and the display seems to properly be using the lookup table for the 3 normal and 1 upside down digit, so that part of the code is also working fine. No incrementing of digits has been observed over hours of runtime.

Pin for pin the CPU is nearly the same. P1.4 has no relay driver on it, and some pullup resistors on the IO to the DS1302 chip are not there.

I could replace the CPU with a known good model, but I can't seem to find the DIP style of STC 15 series available on ebay or anywhere reasonable.

I'm happy to try and debug and share the instructions included with the model to get this working if there is a reason to. Do you have any input on the subject?

Thanks

zerog2k, 2016-08-12:
1. did it come with a schematic and/or board diagram which you can attach here?
2. can you review my board's schematic to see if it has similar hookup:
https://github.com/zerog2k/stc_diyclock/blob/master/docs/DIY_LED_Clock.png
3. does this have the STC15F204EA chip?

zerog2k, 2016-08-12:
Ok, I see that your issue subject indicates the chip is STC15W408AS.

I think this means it has 4k ram, 8k flash, I think W means wide voltage range, i.e. 3.3 or 5v.

Regarding the pullups, if you dont have the strong pullups, then you may need to adjust the pin modes (assuming they are the right pins from mcu to DS1302).
https://github.com/zerog2k/stc_diyclock/blob/master/src/main.c#L187-L188
have a look at section 4 (about gpio pin modes) in stc datasheet (link in reference section of my repo readme). (STC website is super slow... I'm guessing the owner is hosting his site on his cellphone.)

zerog2k, 2016-08-13:
assuming the pinout for STC15W408AS is same as STC15F204EA (looks like it for 28-pin version from internet schematics i can find), and assuming your DS1302 is hooked up to P1.0, P1.1, P1.2, and assuming that lack of pullups might be causing communication issue with DS1302 (time display is pulled directly from RTC), then you could try to put the pin modes into mode 0 (quasi- bi-directional - some pullup):

i.e., try changing:
```
   P1M1 |= (1 << 0) | (1 << 1) | (1 << 2) | (1<<6) | (1<<7);
   P1M0 |= (1 << 0) | (1 << 1) | (1 << 2) | (1<<6) | (1<<7);
```
to
```
   P1M1 |= (1<<6) | (1<<7);
   P1M0 |= (1<<6) | (1<<7);
```

hachi, 2016-08-13:
I think you're on to something. I didn't pay attention to whether the func and + buttons were doing pull up or down the same as your schematic. The mismatch of modes might explain both the RTC comm problem and the lack of buttons.

hachi, 2016-08-13:
You were right regarding the I/O pins for the RTC. The clock portion has started working, but I'm still looking into the broken buttons. They are grounded switches on the same pins as your schematic, so I don't know what's wrong yet.

Thank you for your theory on the 'W' of the part number, the STC web site and Google at large were very unhelpful for finding the data sheet for this chip by the specific part number. After a fair bit of searching I did find that this is the same model of chip with 8k flash. More to come.

zerog2k, 2016-08-13:
Regarding pin modes, mine works fine with default pin mode 0 on DS1302 , so I'll fix this in the code.

Regarding buttons - I think we might be over-thinking the software debounce on them. At least on mine, I hooked up a scope, and there is very little (if any) bounce on key down. There is sometimes some bounce on key up (but usually less than 1ms). So we can probably simplify debounce a bit. I'm also trying to think of how we can better handle long/short press events.

hachi, 2016-08-13:
I think I'm having troubles with timing. The overall speed of the programming is running about double what I would expect after reading the code. The colon flashes about twice per second... maybe 3x.

I raised the delay_ms constant for i to '8' and it looks more correct. However the debounce is also not working and I'm still reading the manual to learn how to adjust the counter constants reasonably.

I've tested with mangled up code and the raw input is working fine, so it's something about the shift algorithm of the debounce keeping the key presses from being detected.

zerog2k, 2016-08-13:
Are you programming w/ STC-ISP or stcgal? What clock speed are you running, is it (around) 11.059 MHz?

zerog2k, 2016-08-13:
Also, I'll be on gitter chat for a few if you want to troubleshoot head-to-head there...
https://gitter.im/zerog2k/stc_diyclock

zerog2k, 2016-08-13:
btw the datasheet for STC15W408AS:
http://www.stcmicro.com/datasheet/STC15F2K60S2-en2.pdf
Better download and save a copy of any STC datasheet - their website can be down or flakey alot.

hachi, 2016-08-14:
To fill in anyone on this reading later...

The STC15W408AS is a slightly faster, similar feature, same pinout CPU of the one this project primarily supports. With the help of @zerog2k there is a confusingly documented lack of Timer1 available on this board and so the button inputs do not work directly.

After a bit of effort I was unable to use Timer2 for the same purpose, but I've been able to get button inputs working by dividing Timer0 and calling to the button handler directly.

Reading from the RTC on my clock which lacked the pullup resistors on the IO and RST lines was accomplished by changing pin modes (applied at https://github.com/zerog2k/stc_diyclock/commit/d5e7f13204cd63bff43b88dd21566d557430445f)

@zerog2k pointed me in the direction of the official STC-ISP program, which seems to be possibly be laden with malware, but documents features of the CPU. It can also generate sample timer and delay loop code with specific timing based on a clock speed.

I've applied the timing fixes necessary, and my hack to cascade from Timer0 to the button handler at https://github.com/hachi/stc_diyclock/commit/8bf437409b88ecebe875ad38f15efff2ecdd68ed

zerog2k, 2016-09-30:
Closing out this issue, as a number of things were improved as a result of this discussion. I have received the STC15W408AS's that hachi donated and confirmed they work with latest master. See release 0.5.0 : https://github.com/zerog2k/stc_diyclock/releases/tag/0.5.0

closed 2016-09-30 by zerog2k