zerog2k/stc_diyclock #13: Swapped pins on external serial connector

jvoosten, 2017-03-12:
While studying the diagram DIY_LED_Clock_STC15W408AS.pdf I noticed the RxD/TxD on the external pins of P1 are swapped. All diagrams of the clocks that I have seen so far connect the 3rd pin of P1 (next to ground) to pin 15 of the controller and the P1/4 to pin 16 of the controller.

zerog2k, 2017-03-13:
@tux75at provide the schematics and has the board, and should be able to confirm this.

tux75at, 2017-03-14:
sorry for the delay....
The orignial schematic is in my repository in the doc folder, possible it went also into your repository.
Maybe i made a mistake, but i will check this as soon as i have time.

tux75at, 2017-03-18:
I fixed the connector P1.
RxD and TxD were accidently swapped. 

I also fixed the transistors. This was something i wanted to look up later, the symbols are a bit strange on KICAD, but when you know this little detail you don't get confused.

Both versions are fixed (KICAD project and PDF). Do i need to make a new pull request?
I am a Github Newbie ;)

zerog2k, 2017-03-18:
Hi @tux75at thanks.
Yes, you can commit them to your fork's master branch, and if that is ready, just create a new PR to my master. When you open the PR, you can mention this issue (e.g. "fixes issue #NN) which would auto-link to this one.

closed 2017-03-18 by jvoosten