zerog2k/stc_diyclock #19: Alternate 7 segment display

JohnOH, 2017-06-30:
I have built the clock kit. I am attempting to load the code onto a stc15w408as processor. The code does not operate as expected. I suspect this is due to the use of a 4 digit display block with two rows of 6 pins. The display appears to be of this general type: http://www.robotshop.com/media/files/pdf/datasheet-com-09483.pdf

Pinout may not be a match. 

I guess that I need to adjust to code to match the display block. I have an unbuilt PCB and will look to buzz out the connections to the processor. I will welcome any suggestions.

zerog2k, 2017-06-30:
you can compare to the existing schematics, which use individually wired 7-segment led modules here:
https://github.com/zerog2k/stc_diyclock/tree/master/docs

if you can produce a schematic or some documentation of how it's wired up, we may be able to create a build-time option which would make a version which works on your hardware.

JohnOH, 2017-07-01:
My display is common anode.

zerog2k, 2017-07-01:
I just create another branch which attempts to separate out the hardware dependent configuration from the code.  May be possible with your setup. Have a look here:
https://github.com/zerog2k/stc_diyclock/blob/hwmodels/src/hwconfig.h

zerog2k, 2017-09-28:
exposed the hardware differences through hwconfig.h which is on master branch

closed 2017-09-28 by zerog2k