zerog2k/stc_diyclock #20: Custom pins support.

MokaSoda, 2017-07-01:
I want to use this firmware because of critical bugs on stock clock kit.
However I cannot change A1~A4 pin numbers of this source code because I don't know the way this code works.
Could you tell me the way to change these pins?
[image]

zerog2k, 2017-07-01:
your schematic has quite a few differences
ntc, ldr, ds1302 all on different pins...
can you scan a better copy of that schematics as pdf or sharper/cleaner jpg? we can add to the docs section of the repo to go along with this "new" model
can you tell me where you got this, or did it have some name?
(I'm trying to think of how to distinguish between the different models... As none of them really seem to have a name, but sometimes use same chip with different layouts, I'm tempted to just call them something unimaginative, like HW_MODEL_A, HW_MODEL_B, etc)

zerog2k, 2017-07-01:
I created a new branch for testing which tries to consolidate and expose all of the various board dependent settings into one file, hwconfig.h:
https://github.com/zerog2k/stc_diyclock/blob/hwmodels/src/hwconfig.h

then, looking at your schematic, I added your board as HW_MODEL_C (because I'm aware of at least two others prior ;)

I compiled a firmware from this branch with the following options:
`make CFLAGS="-D WITH_ALT_LED9 -D WITHOUT_LEDTABLE_RELOC -D HW_MODEL_C"
`
you can try this out and see if it works for you:
[main-HW_MODEL_C.zip](https://github.com/zerog2k/stc_diyclock/files/1117378/main-HW_MODEL_C.zip)

MokaSoda, 2017-07-02:
I bought this from Taobao(Base of Aliexpress). Its name is [ DY-020 ](https://item.taobao.com/item.htm?spm=a230r.1.14.17.ebb2eb2uplIck&id=534676279391&ns=1&abbucket=11#detail) 
Here's more crisp schematics (sorry for upside down images..)
[image]
[image]
After, I upload compiled firmware from you, I notice that clock speed is okay. However, it shows 1F:7F. and the clock minute doesn't count up after a minute. (F is defined in this sourcecode so it should be F)

zerog2k, 2017-07-08:
Can you try setting the clock? Maybe the DS1302 has initial junk values (happens if no power and backup battery is dead)? If this is the case, the process of setting the values should clear this out. If not it may be something else. I just double-checked the hwconfig.h for your model (what we will call "HW_MODEL_C" for now) and it looks correct.

vargham, 2018-02-07:
I have another pinout version with STC15F204EA. I figured out the connections, and I will share it. This version has one 4 digit display. I could also change pin definitions in your code.
[This one from Aliexpress](https://www.aliexpress.com/item/green-color-LED-Module-electronic-clock-LED-microcontroller-SCM-CASE-LED-digital-Alarm-clock-display-time/32646715139.html)

But I have the same problem as above.
When I turn on, it displays 1F 7F and it is never changes.
I can change mode with button 2.
Next mode is temperature, and it is works correctly. I can change C display to F with button 1.
Next mode is unknown for me. It displays an empty digit and then -8-
I tried long and short press combinations with both buttons, but I could not set up the clock.

zerog2k, 2018-02-07:
@vargham I see this has a slightly different 4 digit led module (uses single one rather than several separate led digits) which results in a different scheme required to display properly.
Also other stuff (like ds1302 chip) may have different connections.

Do you have a schematic diagram for this? Likely we need to create another hardware model variant to handle this one. Please open up a new issue so we can track adding support for it.

vargham, 2018-02-07:
> may have different connections
Everything have different connection. It is not a problem, i figured out all the connections. Clock, buttons, led segments, everything. The display is OK, I wrote a little test loop inside your main() and it iterates over all symbols. And your program correctly displays the temperature.

zerog2k, 2018-02-07:
Yes realized that display is working properly if you can see the temperature... Does the kit include schematics? Can you take a good picture or scan and include here? I can abstract any differences in hwconfig.h

zerog2k, 2018-02-07:
@vargham try this:
update these pin mappings to match your ds1302 connections to stc mcu:
https://github.com/zerog2k/stc_diyclock/blob/master/src/hwconfig.h#L84-L89
(note DS_CE is RST pin on ds1302)

Looking at another schematic on aliexpress, it might be something like:

```
 #define DS_CE    P1_1
 #define DS_IO    P1_1
 #define DS_SCLK  P1_0
 // needed for asm optimizations
 #define _DS_IO   _P1_1
 #define _DS_SCLK _P1_0
```

vargham, 2018-02-07:
No, there is no schematic. I have to draw.

> I can abstract any differences in hwconfig.h

I made changes in hwconfig.h with new HW_MODEL_D. Is this occupied? I can change if needed.

zerog2k, 2018-02-07:
yeah - i dont think we have really good names for them as there are so many and they are super generic... HW_MODEL_D is fine ;)

vargham, 2018-02-07:
OK, here is the pinout:

vargham, 2018-02-07:
I will fork, commit. Discuss the error, and pull request if solved. Is It OK?

zerog2k, 2018-02-07:
sure... please put some comments in the hwconfig.h about your new HW_MODEL_D (e.g. link from aliexpress, etc), and I'll take a PR if you can test it out ;)

vargham, 2018-02-08:
https://github.com/vargham/stc_diyclock/issues/1

aFewBits, 2018-02-08:
I probably have schematics and code for that model. Please post a picture of the component side of the circuit board as there are at least three different models. Further differences are that the  LED's can be either  common cathode or common anode and the display LED is all direct drive (with no current limiting resistors) from the I/O ports so you have to diddle the port STC port configurations to increase the current capability of the I/O. The end result is that you end up with somewhat different digit intensities, depending upon how many segments are illuminated. It's a really down and dirty design but they work and they are more compact in size.