zerog2k/stc_diyclock #26: Trickle-charging DS1302

denmmx, 2017-11-19:
Hi, I would like to know, have you disabled in your firmware trickle-charging function of DS1302 RTC chip? In the kit, received from the aliexpress, this feature is enabled, and as a result, it is impossible to use a normal CR1220 battery as it is charging.  MCU is STC15W404AS.

zerog2k, 2017-11-19:
No, in this firmware we do not enable trickle-charge function, which is disabled by default.
Refer to page 7 of maxim's ds1302 datasheet for more info.
https://datasheets.maximintegrated.com/en/ds/DS1302.pdf
I suspect it could be used with rechargeable lion coin cell, nimh, or supercap, but would need to be configured, in the firmware, which currently does not have this feature. (I welcome pull requests if someone wants to add the feature, as a compile-time option.)

denmmx, 2017-11-19:
Thanks for info! 
It is strange that in the original kit this feature was enabled together with using non-rechargeable battery)

zerog2k, 2017-11-19:
curious - how do you know it was enabled? and to what setting?

denmmx, 2017-11-19:
Battery was swollen and that caused the suspicion of the current that flows through it. The multimeter also showed the voltage at the battery pins

zerog2k, 2017-11-19:
interesting... very strange mistake if so... If you were really curioust o know, you could measure the current flowing to the battery, or sniff ds1302 pins on startup to see what they are writing to the trickle charge register (reg write addr: 0x91)
(It's possible the firmware was meant for rechargeable cells or supercaps, but someone assembled the kit using the primary cells...) At the price point, anything is possible ;)

denmmx, 2017-11-19:
it would be interesting... How I can sniff pins of the DS1302? Need some more info :)

zerog2k, 2017-11-19:
Do you have a logic analyzer?

denmmx, 2017-11-21:
Unfortunately have no logic analyzer. I'll just check voltage on battery pins in few days with multimeter, after reinstall stock MCU.

closed 2017-12-16 by zerog2k