zerog2k/stc_diyclock #28: "Waiting for MCU, please cycle power"

tmh00, 2018-01-19:
Apologies if this is the wrong place for this question, I'm new to Git and this is my first attempt to flash an MCU of any kind.

I am working with the Geekcreit clock from banggood: https://www.banggood.com/DIY-4-Digit-LED-Electronic-Clock-Kit-Temperature-Light-Control-Version-p-972289.html

My kit came with an STC15W404AS

I'm using this USB-UART adapter: https://www.ebay.com/itm/CP2102-USB-2-0-to-UART-TTL-6PIN-Module-Serial-Converter-Adapter-Red-Silver-TS/222617611551?epid=1286993754&hash=item33d50b2d1f:g:FqYAAOSwxbpZl5x6

I'm trying to flash my clock from Raspbian Jessie on a Raspberry Pi 3, using STCGAL.  I'm able to build properly, but I'm trying to flash the precompiled <4kb release file, just to be safe.

When I run "make flash," the process is able to detect my adapter, but it hangs on a step saying "Waiting for MCU, please cycle power."  I can't seem to find any way to get past this.  The clock is clearly getting power from the adapter, as it powers on and works as normal when I connect the GND and 5V pins.  I have tried disconnecting and reconnecting the power while at this point, but that doesn't do anything.

Is this expected behavior, and if so, what am I supposed to do to proceed?  If not, what could be wrong?

tmh00, 2018-01-19:
Sorry, I just realized there was a closed thread with the same issue.  However, my solution ended up being much simpler than the other user's.

After a bunch of tinkering, I realized I simply needed to bounce the connection on the GND line at this stage, and not on the +5V line.  I wired up a basic breadboard with a switch on the GND line and I toggle it when I reach the message above.  Works perfectly!

zerog2k, 2018-01-19:
Congrats on resolution. I have wired up the reset pin on mine to date (also adjusted the reset pin options with stcgal) for frequent development.  (Also means I need to open a serial port to activate date to turn the clock on.  I need to investigate simple dtr  pulse, maybe inline cap like Arduino reset uses. )

zerog2k, 2018-01-28:
Yeah dtr pulse through a 1uF cap inline to reset pin works good. I use the stcgal autoreset option "-a".
(assumes you have configured the mcu options to use reset pin as reset.)

closed 2018-01-28 by zerog2k