zerog2k/stc_diyclock #30: Maybe new model? Schematic Provided..

flyingsaucer75, 2018-03-29:
[SKU593344   光控工控YD010(英文）.docx](https://github.com/zerog2k/stc_diyclock/files/1858988/SKU593344.YD010.docx)

I've one of this could you please tell me if it's supported?
(Led Digit have Amber colors stock firmware has too dark light when ambient is full dark I must always set to disable automatic dim)

P.S.
I'm only windows user could you please provide a guide to compile and flash?
Command lines etc..

Best regards
Tony

aFewBits, 2018-03-30:
I had the same problem with my yellow color unit.

I exchanged the (8) 510 ohm segment resistors with 47 ohm parts. This allows for a nice brightness range. While this appears to be a large change, the resulting current was within expected limits. I can only surmise the yellow LED's are significantly dimmer than the other colors.

flyingsaucer75, 2018-03-30:
@aFewBits
First all thanks a lots for your work, I've perfectly repaired another Clock kkmoon with your source.
I've installed kali linux in a virtual machine installed sdcc and compiled modifying global.h, then I've programmed the chip by windows with a little usb to ttl based on cp2102 using stc-isp-15xx-v6.86L.exe.
Now I've another one I do not succeed in compiling it correctly it gives always some error.
This is the schematic:

[image]

With this one I've tried with this changes:

flyingsaucer75, 2018-03-30:
Closed issue by error sorry people.

aFewBits, 2018-03-31:
With zerog2k's firmware, IIRC, the 1F:7F display occurs when DS1302 is not communicating with the processor. I just posted a fix for the 15F204EA processor type on my repo and if you have comments or questions on my code, you should ask a question on that repo rather than here.

flyingsaucer75, 2018-03-31:
aFewBits you are right I'll close here and open issue there. Only for info also my model something hybrid with MODEL_C (it misses only relay on PCB) needs:

Changed DS1302's CE to open-drain output

    //M1 M0
    // 0 0 Bidirectional
    // 0 1 Push-Pull
    // 1 0 Input
    // 1 1 Open-drain
    //          IO CE CK
    P1M1 = 0b00000010;
    P1M0 = 0b00000010;

Constant 1F 7F has been changed to constant 00 00, but still could not set time.
Maybe all DS1302 pins need to be set to bidirectional as vargham suggested for his Model_D and solved.
I don't know how to do it.
@aFewBits thanks for helping opening question on your repo now.

flyingsaucer75, 2018-03-31:
Maybe better leave this open to permit also zerog2k to add all DS1302 pins bidirectional for model_c

zerog2k, 2018-04-06:
As far as I understand it, from page 54 (5.1 Special Function Registers) of the STC15F204EA datasheet,  power-on defaults for P0Mx and P1Mx registers should be all zeroes, meaning pinmode is "quasi-bidirectional". I have seen this issue (ds1302 communication) reported several places, but myself have not been able to reproduce it. All of the models I had are slightly different w.r.t. pullups on various ds1302 lines, and different from what folks are having... I think the manufacturers keep cutting corners and  saving $0.0001 on external pullup resistors here and there..
I think a "real ds1302" is supposed to have something like approx 40k pull_down_ resistor on the signal lines, fwiw.

For this particular model, you may need to add a new model type if this is unlike anything existing. It sounds like it might be the case.

Madhava-04, 2018-06-29:
@flyingsaucer75 
You referred in post date Mar 30, that green color clock with afewbits compiled source now is working perfectly.  Is it possible to upload HEX file of that ?  I have no knowledge of writing or compiling codes.  I want to try this HEX on my broken clock ! I have a clock with RED Leds and I attach circuit diag[
[image]
](url)