zerog2k/stc_diyclock #31: 4 digit display models?

Hegz, 2018-04-08:
Hi, I purchased this unit:

https://www.ebay.ca/itm/New-LED-DIY-Digital-Electronic-Micro-Controller-Kit-Clock-Time-Thermometer-CA-/292048065603?var=&hash=item899c6ccd6d

The major difference that I can see would be that this unit uses a single 4 digit seven segment display. I'm not sure if this firmware would have support for that.

Schematic:
[clock.pdf](https://github.com/zerog2k/stc_diyclock/files/1887038/clock.pdf)

Will this work with the open source firmware?  It would be fantastic to stop this thing changing the display, any maybe turn down the LED intensity...

vargham, 2018-04-09:
I have the same model. I made a working fork for that model. Check here: https://github.com/vargham/stc_diyclock

zerog2k, 2018-04-13:
Thanks for contributing @vargham! Welcome and PRs back upstream...