zerog2k/stc_diyclock #37: синхронизация времени с приемником GPS вывод последовательных данных NMEA

nemo88888, 2019-05-05:
Hi!
Synchronization of time with the GPS receiver conclusion of consecutive data of NMEA.
Prompt what GPS receiver to use and where to connect on the scheme of hours?

zerog2k, 2019-05-05:
Hi, I don't really understand your question.
However, there is _experimental_ gps support on the `gps` branch.

To test, I hooked up NMEA GPS module like this one
https://www.banggood.com/GPS-Module-for-Auto-Car-DVR-Navigator-Tracking-Device-Recording-Car-Dash-Camera-p-1049257.html?p=WX0407753399201409DA
which outputs serial nmea at either 9600 bps into P3_6:rxd (look at `uart.h`)

Also, take a look at `gps.h`, and see that it is currently looking for `$GPRMC` strings, however if you have GLONASS, you may need to modify this to `$GLRMC` (I'm in USA, I have only ever seen $GP strings.)

nemo88888, 2019-05-05:
It is clear.

This experimental program can approach by these o'clock? 
https://ru.aliexpress.com/item/DIY-Kit/32903754608.html
If is not present, then what needs to be changed?

And I cannot find hours on the STC15W408AS microcontroller yet. Specify the reference if you know.

zerog2k, 2019-05-06:
It may be difficult to tell from the pictures. Unfortunately even in that listing:
1. schematic shows both STCW404AS and STCW408AS as mcu ;)
2. pictures show STC15F204EA

STCW408AS is best choice as it has 8KB flash, and hardware UART peripheral.
STCW404AS _might_ work, if you can either get code size down to <4KB _or_ use tricks like EEPROM space (9KB) for storing large constants and look-up tables.

Probably will _not_ work with STC15F204EA (no hardware UART peripheral) and only 4KB of code space, and 1KB EEPROM.

nemo88888, 2019-05-06:
And at you with what microcontroller synchronization of time with the GPS receiver is used? 
An exit from the GPS TX or PPS receiver?

zerog2k, 2019-05-09:
> And at you with what microcontroller synchronization of time with the GPS receiver is used?
> An exit from the GPS TX or PPS receiver?

I don't exactly understand what you are asking.
Yes, I am parsing the NMEA output strings from GPS TX, particularly `RMC` sentence.
https://www.gpsinformation.org/dale/nmea.htm#RMC
(I did look at `ZDA` however, some gps receivers did not seem to provide this. `RMC` seems to have what I need).

nemo88888, 2019-05-09:
Through online the translator it is difficult to communicate. If I correctly understand, the speech in your article goes about synchronization of the electronic clock on the STC15W408AS or STC15F204EA processor with the GPS receiver. I wanted to learn from you whether it is possible to use your pilot project for the electronic clock on processor 15W404AS (it is used at me in the electronic clock) and where to connect the GPS receiver in the scheme of hours. I plan to use this receiver

https://ru.aliexpress.com/item/1811853522.html?productId=1811853522&productSubject=2pcs-lot-GY-NEO6MV2-new-GPS-module-with-Flight-Control-Flight-Control-EEPROM-MWC-APM2-5&af=506321&aff_platform=aaf&cpt=1556717407284&afref=https%253A%252F%252Fpafutos.com%252F&dp=5a44cc0233724e66173c22d8dd025695&onelink_item_from=1811853522&onelink_thrd=0.0&onelink_page_from=ITEM_DETAIL&cv=47843&onelink_item_to=1811853522&pvid=440653f3-527f-42aa-aff5-fc0c7ca2cfaa&mall_affr=pr3&onelink_duration=0.704075&sk=VnYZvQVf&aff_trace_key=459ae4f3dc8a49f19aff30a2cda40390-1556717407284-00579-VnYZvQVf&onelink_status=noneresult&scm=1007.22893.125764.0&terminal_id=cc16b801dfa144a0805022164fcb5f08&onelink_page_to=ITEM_DETAIL

I do not develop the program for microcontrollers. Did not find articles of similar to this yours in the Internet. Therefore wanted to obtain exact information on the questions interesting me.

zerog2k, 2019-05-12:
It should be possible with STC15W404AS - this is basically same mcu as STC15W408AS. Difference is 404 has 4k code space + 9k eeprom, 408 has 8k code + 5k eeprom.
Also both have hardware uart peripheral. STC15F204EA does not have hardware uart, so that along with less overall code space, would be difficult (but not impossible) to support GPS decoding.

On either STC15W404AS or STC15W408AS, both support hardware remapping of the UART1 to P3_6:rxd, P3_7:txd.

So to support any serial/uart gps module you need to know a few things to adjust firmware to support it.
1. need to know output format. Should be NMEA.
2. need to know serial baud rate. Probably either 9600 or 4800 bps.
3. need to know what kind of NMEA "sentences" it outputs, e.g. `$GPRMC...` or `$GLRMC...` or others

nemo88888, 2019-10-26:
Привет!
Снова я к Вам, но с конкретной просьбой. Вы можете изменить прошивку для STC15W404AS, чтобы функциональность была такая:
* 55 секунд отображается текущее время в формате 24 часа
* 5 секунд отображается температура только в градусах Цельсия
* ночью, когда освещение падает, яркость часов автоматически уменьшается (как в стандартной прошивке с датчиком освещения, идущим в комплекте)
* простой будильник (согласно стандартной прошивке, но без почасового сигнала)
* GPS (модуль GY-NEO6MV2):
 1. стандартные данные NMEA
 2. скорость 9600 bps
 3. выводит GNRMC и GNGGA
Всё остальное, согласно стандартной прошивке, не нужно.