zerog2k/stc_diyclock #40: Привет!

nemo88888, 2019-10-26:
Again I am to you, but with a specific request. You can change the firmware so STC15W404AS the functionality is:

55 seconds displays the current time in 24 hours format
5 seconds displays temperature in degrees Celsius only
At night, when the lighting drops, the clock brightness automatically decreases (as in standard firmware with the lighting sensor coming in)
Simple alarm clock (according to standard firmware, but without hourly signal)
GPS (GY-NEO6MV2 module):
Standard NMEA data
Speed 9600 bps
Displays GNRMC and GNGGA
Everything else, according to standard firmware, is unnecessary.

zerog2k, 2019-10-26:
Sure, I will review any pull requests which have these functionalities.
I suggest making these options as compiler defines, such that they can be "configurable" from the hardware config header. This will make it easier for users to decide which features they want enabled or disabled when compiling.

nemo88888, 2019-10-27:
Привет!
Полностью с Вами согласен в этом, чтобы пользователи сами могли выбрать какие функции они хотят включить или отключить при компиляции.

nemo88888, 2019-11-27:
Привет!
Вы не занимались моей просьбой?
Вот что нашел в интернете - если соединить вывод 7 и GND, часы будут показывать 55 секунд время и 5 секунд температуру.
Поэтому задача упростится. Только прием GPS для коррекции часов со спутника.
Можно ли сделать отдельную кнопку для коррекции часов? Если минуты часов отстают, то при нажатии на кнопку коррекции, минуты обнуляются с увеличением часа, если минуты спешат, то при нажатии кнопки коррекции минуты обнуляются, но увеличение часа не происходит.