zerog2k/stc_diyclock #41: Dot-Matrix version?

MarqueIV, 2019-11-07:
Hi! Apparently there’s a new dotmatrix version of these little clocks, which would also make it really easy to display day of the week as well as have a true menu for the settings.

Here’s a link to the product on Amazon:
Amzn.com/B07T369S58

No idea where to track down the original code, but I’m wondering if yours can be modified to work with this. I would be more than happy to contribute to the cause and ship you one of these clocks if you wanted to take a stab at it. Heck… even if you can just get a simple hello world on the screen, I could probably take it from there. I’ve already soldered on the header on the circuitboard so I should be good to go with programming. I just have no idea where to start.

zerog2k, 2019-11-08:
Hi MarquelV, I think it's very possible to code for these, if the one you are lookign at uses STC microcontroller.
Please email me directly at `af5mi` at `yahoo` dot `com` about contributing a hardware sample.

-Jens

zerog2k, 2019-12-16:
hi @MarqueIV were you able to test out https://github.com/zerog2k/clock-3118AY/tree/testing ?

MarqueIV, 2019-12-19:
Crap! Completely forgot about this!  Lemme see if I can take a crack at this in the next week or so.  Sorry for the delay.