zerog2k/stc_diyclock #42: hourly chime

venustrg, 2019-11-23:
i like to have a hourly chime in my clock but not so annoying as in original firmware, just a short beep. also i like functionality and logic of your firmware. so here is the patch for hourly chime support. chime display are right after alarm, both displays are with timeout when not adjusting. 12h/24h supported, dot1/dot3 are pm indicators for start/stop hours, dot2 - chime on/off, shown only at chime display.

works fine with my STC15W404AS.

of course, chime can be disabled with WITHOUT_CHIME at compile time.

[diyclock_chime_1224.diff.txt](https://github.com/zerog2k/stc_diyclock/files/3882576/diyclock_chime_1224.diff.txt)

zerog2k, 2019-11-23:
hi @venustrg, please submit any patches as a pull-request
thanks

venustrg, 2019-11-25:
did so

zerog2k, 2019-12-16:
fixed with #43

closed 2019-12-16 by zerog2k