zerog2k/stc_diyclock #44: Weird bevaviour with STC15W408AS

Kobbe1, 2020-02-01:
Hi there,

I have what I believe is one of the banggood kits. I did successfully flash this frimware a while ago, I can't quite remember what settings though.
Recently I fried the µ-controller while trying to flash - I think by connecting it incorrectly.

I therefore replaced the micro - however not  with the STC15F204EA as originally but with an STC15W408AS I thought to be pin compatible.

I played with the compuile options - but I only get the result as attached.
The right most digit goes on/off with light level.

I swapped the segment port and digit and interestingly the pattern changes completely, telling me that the micro is indeed driving the segments somehow.

Any ideas? Maybe pullups one of the µ-controllers has and the other does not?

Thanks a lot,

-Kobbe
[image]
[image]

zerog2k, 2020-02-02:
This looks different from the ones this firmware is for.
This firmware is for variants which drive individual 7 segment digits. Yours looks like maybe it has a 4x 7-segment display driven as a single unit, which probably has slightly different pinout.

The STC15F204EA and STC15W404AS/408AS are essentially same pinout, just the others have more ram, code, and peripherals

Do you have a schematic?

zerog2k, 2020-02-02:
If you are sure it was working before with this firmware, perhaps it's one of the variants, e.g. `HW_MODEL_C`
review this issue: https://github.com/zerog2k/stc_diyclock/issues/20
and if it is, likely you could try building it with this option added to platformio.ini:
`build_flags = -D HW_MODEL_C`

Kobbe1, 2020-02-10:
Sorry for the delay but wanted to try it again to be sure. I had played around with the DHW_MODEL_C flag in the makefile before, tried it again, but 0 change.
The only thing that makes a difference is to swap digit and segment port in the code - still not readable but the readout changes.
Just in case below is a log of the last build in case you spot any dumb mistakes ...

`:~/stc_diyclock$ make
sdcc -o build/ src/main.c --code-size 4089 --xram-size 0 --data-loc 0x30 --disable-warning 126 --disable-warning 59 -Dstc15f204ea -DWITH_ALT_LED9 -DWITHOUT_LEDTABLE_RELOC -DSHOW_TEMP_DATE_WEEKDAY -DHW_MODEL_C build/adc.rel build/ds1302.rel
   Name             Start    End      Size     Max     
   ---------------- -------- -------- -------- --------
   ROM/EPROM/FLASH  0x0000   0x0dff    3584     4089   
cp build/main.ihx main.hex
tk@tk-TravelMate-2350:~/stc_diyclock$ STCGALPROT="stc15" make flash
stcgal/stcgal.py -p /dev/ttyUSB0 -P stc15 -t 11059  main.hex
Waiting for MCU, please cycle power: done
Target model:
  Name: STC15W408AS
  Magic: F51F
  Code flash: 8.0 KB
  EEPROM flash: 5.0 KB
Target frequency: 11.075 MHz
Target BSL version: 7.2.5T
Target wakeup frequency: 36.600 KHz
Target options:
  reset_pin_enabled=False
  clock_source=internal
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=3
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
Loading flash: 3584 bytes (Intel HEX)
Trimming frequency: 11.057 MHz
Switching to 19200 baud: done
Erasing flash: done
Writing flash: 3648 Bytes [00:03, 1038.72 Bytes/s]                                                                           
Finishing write: done
Setting options: done
Target UID: F51F44B103089C
Disconnected!
`

zerog2k, 2020-02-10:
Can you share schematics? It’s the only way I can see what is different.

Kobbe1, 2020-02-19:
I have another one coming from banggood, will scan schematic once I have it. Is there a way to backup the original firmware or is this locked?

zerog2k, 2020-02-20:
No, no known way to readout flash - the STC uart bootloader rom does not support reading (by design).

Kobbe1, 2020-05-10:
Hi there,
the new kit arrived.
Guess what - it is a different hardware altogether. I will close this issue here, the old clock has a hardware issue I am convinced now.

I will open a new one as also the new kit gives me trouble, this clock drives me crazy :)

closed 2020-05-10 by Kobbe1