zerog2k/stc_diyclock #61: Firmware for clock with voice

icom512, 2021-12-17:
[image]

Will any of the firmware work for this clock? Can someone make the .hex file that I need.

Thank you

zerog2k, 2021-12-18:
You can compare the schematic to existing known variants from hwconfig.h file. If it's not one of the ones supported, it should not be too hard to add support for it, as it looks similar to others.
The only thing new would be the voice/sound ic, which current fw has no provisions for. It can be added I'm sure if someone wants to try adding support for it.

icom512, 2021-12-18:
Thank you zerog2k for answering me. I have no ideal about programing. I have look at the code to try to understand it, but it is way over my head. If the voice/sound does not work that is no big deal. Could you build a hex file for me? I would really appreciate it. 

Thank you.

SYSirkent, 2023-02-27:
Hello, when I was looking for materials for my kit, I came across information about the above OTP.
Here is the link: https://www.nyquest.com.tw/en/product/SpeechIC/NY3_Series/download
Hope that helps.