zerog2k/stc_diyclock #62: Worked for 10 minutes with new chips and then stopped...

alexanderre, 2021-12-17:
So I tried anything I possibly could with stock chips - no luck. Ordered a set of new both STC15W404AS and DS1302 - actually like 2 of STC and 10 of DS1302. Put a new set in, flashed STC - worked for like 10 minutes perfectly well and then stopped. I disassembled the clock again and found that DS1302 gets nasty hot - I actually slightly burned my finger.

Any ideas on why is that possible? Anything I could do to fix that? 

Thanks in advance and great job!

zerog2k, 2021-12-18:
ds1302 getting hot, I suspect either something wrong with the wiring/assembly of the board, or it's some hardware variant which has different DS1302 to MCU connections than any of the current ones in software. Do you have a schematic of your unit? Can you compare it to the variants in hwconfig.h

alexanderre, 2022-01-23:
> ds1302 getting hot, I suspect either something wrong with the wiring/assembly of the board, or it's some hardware variant which has different DS1302 to MCU connections than any of the current ones in software. Do you have a schematic of your unit? Can you compare it to the variants in hwconfig.h

Now I know what was the problem - i used a bad flux. It turned out to be too active and even after multiple washes it stayed under the ICs and corroded the contact joints. I bought assembled another similar kit and now it's perfect 🙌🏻. 

P.S.: is there a way to retrieve an original firmware from the supplied chip? Just as a backup. I have a USB-TTL cable and all the necessary software, but a hint would be greatly appreciated

zerog2k, 2022-01-23:
> is there a way to retrieve an original firmware from the supplied chip?

There is no known way of retrieving firmware from the STC chips. It's intentionally a programming only uart interface.

closed 2022-01-23 by zerog2k