zerog2k/stc_diyclock #66: Random slow downs. STC15W408AS

p23f, 2023-03-15:
I have built three of these clocks and they are all having the same issue with this latest firmware.  My clocks have the STCW408AS chip.

Everything starts off fine then...

At random times, usually under a day the display will slow down (as in instead of 60 seconds to change to the next minute it takes a lot longer) but the flashing two dots still stay the same, flashing every second.   If I bring the seconds display up, the dots will flash maybe ten times, and then the second digit will turn to the next number.   Then some time later that will change to maybe 6 flashes before the seconds digit changes, then maybe 8 flashes.  Then out of the blue it will go back to normal.

I have swapped chips between devices and it makes no difference.

p23f, 2023-03-23:
You can ignore my above comment.   I bought four of these units (one, then three afterwards).  I have built three of them and they all would lock up at random times, swapping chips made no difference.  I tried different firmwares as well.

Now all them are acting normally with this firmware.

zerog2k, 2023-03-29:
I can only suggest carefully inspecting the pcb and assembly. Things like a shorted lead, or poor crystal and/or RTC IC connections could cause this kinds of issues.